jesseweisberg/moveo_ros #1: Problem loading Moveo 3D model

alequeiroz, 2017-12-23:
Hello guys, i was following the ROs instructions to install it, and i droped the moveo_ros folder into src folder of my catkin and run _$ roslaunch moveo_moveit_config demo.launch
_ than it give me a few erros but the program loads but without the 3d model, maybe i installed it wrong?
Here is the erros given:
`NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    move_group (moveit_ros_move_group/move_group)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz_alessandro_31232_3816467714757476022 (rviz/rviz)
    virtual_joint_broadcaster_0 (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[virtual_joint_broadcaster_0-1]: started with pid [31250]
process[joint_state_publisher-2]: started with pid [31251]
process[robot_state_publisher-3]: started with pid [31258]
ERROR: cannot launch node of type [moveit_ros_move_group/move_group]: moveit_ros_move_group
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/alessandro/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
process[rviz_alessandro_31232_3816467714757476022-5]: started with pid [31263]
[ WARN] [1513989752.190574867]: Shutdown request received.
[ WARN] [1513989752.190640317]: Reason given for shutdown: [new node registered with same name]
_IceTransSocketUNIXConnect: Cannot connect to non-local host alessandro-Inspiron-5548
_IceTransSocketUNIXConnect: Cannot connect to non-local host alessandro-Inspiron-5548
Qt: Session management error: Could not open network socket
[ INFO] [1513989752.283390458]: rviz version 1.12.13
[ INFO] [1513989752.283488549]: compiled against Qt version 5.5.1
[ INFO] [1513989752.283523453]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1513989752.409755186]: Stereo is NOT SUPPORTED
[ INFO] [1513989752.409837214]: OpenGl version: 3 (GLSL 1.3).
[ERROR] [1513989752.474188894]: PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load.  Error: According to the loaded plugin descriptions the class moveit_rviz_plugin/MotionPlanning with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu
[ERROR] [1513989752.493747032]: PluginlibFactory: The plugin for class 'rviz_visual_tools/RvizVisualToolsGui' failed to load.  Error: According to the loaded plugin descriptions the class rviz_visual_tools/RvizVisualToolsGui with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
`

jesseweisberg, 2017-12-24:
Have you downloaded Moveit? If you run roscd moveit_ros_move_group, does it find the package?

If not, download instructions here: http://moveit.ros.org/install/

alequeiroz, 2018-01-29:
It was not installed, after installed it worked, but the real time object is not working, after reinstalled open cv 3.0 when i trie to build the conda enviroment it give me the error:  
moveo_ros/object_detector_app$ conda env create -f environment.yml
Using Anaconda API: https://api.anaconda.org
Solving environment: failed

ResolvePackageNotFound: 
  - opencv3==3.0.0=py35_0
  - tbb==4.3_20141023=0

is it normal?

jesseweisberg, 2018-03-11:
I actually did not use Anaconda, but after some research it seems like something related to the specific versions between opencv3 and Anaconda.  The purpose of using Anaconda I believe is just to create a virtual environment, and for that I just used virtualenv.  

[This is a good tutorial](https://www.pyimagesearch.com/2017/09/25/configuring-ubuntu-for-deep-learning-with-python/) on how to set up your virtualenv with all the necessary requirements (Python 3.5, OpenCV 3.0, Tensorflow 1.2).

After you've created your virtualenv with all the necessary requirements, you should be able to run the object detection app (`python object_detection_app.py`).

jesseweisberg, 2018-08-03:
Closed due to inactivity.

closed 2018-08-03 by jesseweisberg