jesseweisberg/moveo_ros #3: catkin_make error

wheemzy1, 2018-02-07:
Hi Jesse,

Thanks for contributing this. I'm just getting familiar with ROS. After I installed Moveit and placed you package files in my catkin_workspace/src I then attempted to build using catkin_make. I received this error. 
......
[ 81%] Building CXX object moveo_ros-master/moveo_moveit/CMakeFiles/move_group_1.dir/src/move_group_interface_coor_1.cpp.o
[ 90%] Built target moveo_moveit_generate_messages
In file included from /home/matthew/catkin_ws/src/moveo_ros-master/moveo_moveit/src/move_group_interface_coor_1.cpp:1:0:
/opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h:76:20: error: expected ‘;’ at end of member declaration
    operator bool() const
                    ^
moveo_ros-master/moveo_moveit/CMakeFiles/move_group_1.dir/build.make:62: recipe for target 'moveo_ros-master/moveo_moveit/CMakeFiles/move_group_1.dir/src/move_group_interface_coor_1.cpp.o' failed
make[2]: *** [moveo_ros-master/moveo_moveit/CMakeFiles/move_group_1.dir/src/move_group_interface_coor_1.cpp.o] Error 1
CMakeFiles/Makefile2:1613: recipe for target 'moveo_ros-master/moveo_moveit/CMakeFiles/move_group_1.dir/all' failed
make[1]: *** [moveo_ros-master/moveo_moveit/CMakeFiles/move_group_1.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I tried to troubleshoot and the closest I could find is this:  

https://github.com/ros-planning/moveit/commit/4033e577b64ab7ef0e4473342168494f69ef9324#diff-c41191af5080252e849ad41db505eac3

I tried modifying the first file listed, but could not locate the 2nd and 3rd files in order to modify. 
When I ran again, I still received an error. 

Please help.

nez, 2018-02-08:
Hi @wheemzy1, I tried the following and now I'm being able to build:

on moveo_moveit/src/move_group_interface_coor_1.cpp on line 126:
remove:
```bool success = move_group.plan(my_plan);

ROS_INFO_NAMED("moveo", "Visualizing plan 1 (pose goal) %s", success ? "" : "FAILED");```

and replace with:
```move_group.plan(my_plan);```

I tried this, because move_group.plan returns apparently a MoveItErrorCode instead of boolean. (I don't know yet how to catch the status code correctly)

wheemzy1, 2018-02-08:
thanks Nez, that solved it

closed 2018-02-08 by wheemzy1

wheemzy1, 2018-02-08:
https://github.com/ros-planning/moveit/issues/759

nez, 2018-02-08:
Hi @wheemzy1 great, so we can implement the solution offered by @BryceStevenWilley are you gonna make the PR? Otherwise I will :)

wheemzy1, 2018-02-09:
go for it bud.