jesseweisberg/moveo_ros #4: "error: moveo_moveit/ArmJointState.h: No such file or directory" persists

nez, 2018-02-08:
Hi @jesseweisberg the error persists even though I have removed the ros_lib folder.

I'm using kinetic on 16.04, and have installed all moveit packages. Maybe I'm missing a package, or maybe I can copy moveo_moveit/ArmJointState.h somewhere?

jesseweisberg, 2018-02-08:
Hi @nez!  After you remove the ros_lib folder, there is one command left to generate all the necessary libraries (including moveo_moveit/ArmJointState.h).  Make sure not to forget the "." at the end of the last command below.
```
cd <Arduino sketchbook>/libraries
 rm -rf ros_lib 
 rosrun rosserial_arduino make_libraries.py .
```
Let me know if this solves the problem.

nez, 2018-02-09:
Thanks @jesseweisberg I could make it work by putting the libraries in the ~/Arduino/libraries folder, thanks for your help! :)

closed 2018-02-09 by nez

rootrootdoul, 2018-12-15:
i did it all and still get the same error can somoene help me out?