jesseweisberg/moveo_ros #6: Can not launch the program

fifo152, 2018-02-17:
I have got a problem, when I try to run the program I always encounter a problem: 
[demo.launch] is neither a launch file in package [moveo_moveit_config] nor is [moveo_moveit_config] a launch file name

As far as I know I have everything installed correctly even that I have never used linux before. 
I am trying to run it on ubuntu 16.04

jesseweisberg, 2018-03-11:
It may be helpful to run through the initial ROS tutorials if you haven't already.  Check out [this answer](https://answers.ros.org/question/143496/roslaunch-is-neither-a-launch-file-in-package-nor-is-a-launch-file-name/) on the ROS site.  It seems like the launch file and the package are not being recognized. 

Did you source your catkin workspace before calling roslaunch?  You can do this by executing the following command.
`source ~/.../catkin_ws/devel/setup.bash`
Then, try:
`roslaunch moveo_moveit_config demo.launch`

jesseweisberg, 2018-05-10:
closed due to inactivity

closed 2018-05-10 by jesseweisberg