jesseweisberg/moveo_ros #7: sketch upload, ArmJointState.h error

bjela, 2018-03-01:
As mentioned, compiling error occurs. 

Yet after several library resets and another shot on different machine i'm sure I do not understand the process. 
Please advise, not a ROS rockstar

jesseweisberg, 2018-03-11:
Please give more information about your issue so I can try to help.

jesseweisberg, 2018-03-11:
Perhaps [this previous issue thread](https://github.com/jesseweisberg/moveo_ros/issues/4) may shed some light on your current problem.

Jeroenh89, 2018-03-24:
Hi Jesse,

Iam currently havingthe same problem. Iam running Ubuntu 16.04 with Kinetic. My moveo_ros file is in the src directory.  I have the Moveit package succesfully running and i can do some fancy plan and execute simulations. 

I have the Arduino - ros lib unpacked in the libraries directory of arduino. I can do some simple arduino blinking with the example programs. 

However, when compiling the moveo_moveit_arduino i get the same error. Cant locate the "moveo_moveit/Armjointstate.h." 

I've deleted the ros_lib directory and used the "  rosrun rosserial_arduino make_libraries.py .
" command in the arduino - libraries folder. However, it does not seem to generate the armjointstate.h file in the arduino ros_lib.

Jeroenh89, 2018-03-25:
Hmm, the catkin_make file does generate the ArmJointState.h in the build folder. However, if i run the "rosrun rosserial_arduino make_libraries.py . "  the file is not added to the ros_lib.

Jeroenh89, 2018-03-25:
found the problem! I was running the source setup.bash and the rosrun rosserial_arduino make libraries in two differtent terminals... #noobmistake 

Now it seems to work!

bjela, 2018-05-10:
Sweed Jeebus, that worked for me as well
Thanks guys!

closed 2018-05-10 by jesseweisberg

bjela, 2018-05-15:
Might be worth noting that sketch should be loaded from the catkin folder

advait0810, 2018-07-04:
Hello. I've tried everything mentioned here and other places. My setup.bash files in kinetic and devel have been sourced. Everything in the same terminal as well. But it's not working.

advait0810, 2018-07-04:
@ghost , what do you meant exactly by "sketch should be loaded from the catkin folder"?

rootrootdoul, 2018-12-16:
still not working