jesseweisberg/moveo_ros #8: Wiring

alexanderwatanabe, 2018-03-08:
Hello! First off I love your project. I'm just about finished with my Moveo (trouble sourcing nema14's where I live, may redesign last 2 axis's for nema17) but I have the first 4 axis's up and running using pronterface.

What's not clear to me from looking at your arduino sketch as well as the wiring diagram here ( https://github.com/jesseweisberg/moveo_ros/blob/master/aerial_robot_sketch.png ) is how the dual steppers on joint 2 are accounted for. Am I supposed to wire up a transistor on the DIR logic pin so that when one stepper driver gets a HIGH signal, the other is low? Or do we slave another axis with inverted logic? Or something else?

Thanks,

Alexander

alexanderwatanabe, 2018-03-08:
Just to clarify - I do see that in the diagram the X axis on the ramps 1.4 seems to be split to two steppers - I guess you are wiring them up on the stepper side of the driver to rotate in opposite directions?

jesseweisberg, 2018-03-11:
Hi Alexander– thanks for pointing this out, because I forgot to mention what I did there.   I did exactly what you mentioned in the second comment.  On one of the Joint 2 motors I flipped the B+ and B- terminals on the TB6560, and if you set it up the same way I did, that should solve your problem.

closed 2018-03-11 by jesseweisberg