jesseweisberg/moveo_ros #10: Jerky stepper movement with trajectory planning

Jeroenh89, 2018-03-30:
Hi,

First of all, thanks for sharing this ROS stack! it really helps me alot with my robot arm. 

But Whenever i use the trajectory planning, my stepper motors are moving a bit jerky at lower speeds. See this video: 

https://www.youtube.com/watch?v=EUeTrV3aFAA

if i use the terminal to move the joints with "rostopic pub joint_steps moveo_moveit/ArmJointState <Joint1 Joint2 Joint3 Joint4 Joint5 0>" 

https://www.youtube.com/watch?v=MEVdPdOFKVA

Both are with the same max speed. 

I have the following setup:

Ubuntu 16 04 running Kenetic
Arduino Due 
Leadshine DM556 on 1/16th microstepping and TMC2208 for the smaller motors.

Any suggestions?

Thanks!

jesseweisberg, 2018-05-10:
Hi, thanks for the well documented issue!  I apologize for the late response. If you perform the exact same trajectory with the same max speed from both terminal and trajectory planning, is the trajectory-planned motion still jerkier?  

I appreciate the videos!  However, it may help us more to crack the bug if we have videos or just check when both trajectories are identical.  

I will be quicker with responses from here on out.

Jeroenh89, 2018-05-12:
In the meantime I’ve traced the problem back to "moveo_moveit moveit_convert" script.  The output of this script is about 10 hz, measured via rostopic hz.  In the script however, the loop rate is set to 20 Hz see:  "ros::Rate loop_rate(20);"

For some reason, the script does not run faster than 10 Hz, and if lower it to 5 or 1 Hz, the output of the script does respond to it, and I notice more "lag / jerky movement" on the robot arm. 
I think that the output of 10 Hz is a bit too slow for the current setup and the jerky movement comes from not receiving new positions fast enough. 

I'm trying to figuring out why the speed of the script is capped to 10 Hz. Do you have any suggestions? Wat is the output of your script?

jesseweisberg, 2018-05-21:
I think you're onto something here.  My robot isn't super jerky, but I do notice the following when I run rostopic hz /joint_steps:
1) The first movements run at about 10hz
2) The frequency drops off pretty quickly to around 1-3hz.  I think there is some performance dropoff here as well.

I'm not sure exactly what the bottleneck is in the script.  I know in the past I was having buffer issues with rosserial because I was sending too much information across at a time, but I don't know if that's the issue now.   It very well may be the way the script or Arduino firmware is written (I really just got it to work and haven't really worked on perfecting it since).

I'll look further into it when I get a chance and let you know if I find anything.

pranavb104, 2018-12-05:
Im also having similar problems , where my robot has jerky movements and sometimes it is not completing it's motion. I also suspect the problems  is in the ros publish rate or the delay in the arduino loop. But still no permanant solution.

Did you guys find a fix for this? Please let me know...

Jeroenh89, 2019-02-20:
Not yet!

kdraun, 2020-05-01:
I think it has to do with the way the joint_states data is received by the arduino. As the signal is calculated, and each of the steppers carries out the steps they need to carry out at each timestamp, some of the steppers will eventually finish before others, as they have less steps to carry out. They will then stop the motion, while others will continue during the timestep. 
Next timestamp, they will start again, leading to jerky motion.
Solution to this is to spread the steps to be carried out, evenly in each of the timestamp.

nez, 2021-07-14:
@Jeroenh89 btw, nice robot, what design are you using?

nez, 2021-08-18:
I had the same issue when using moveit with the default settings, I could solve the issue setting the speed multipliers from 0.1 to 1