jesseweisberg/moveo_ros #11: Stepper power not supplied

bjela, 2018-05-15:
Pins are set correctly, but like they are not enabled. 
It's mega with RAMPS1.4 
Motors are controllable over marlin, but not trough moveo_ros

I tried supplying power to Enable pin to the driver separately, seems that is not the issue. 

Do you have any ideas I might have overseen?

Best,
B

Jeroenh89, 2018-05-15:
What kind of drivers are you using? 

It could be that you need to add a line to the arduino code to make a pin (like standby) of your driver actively high or low, instead of floating. 

Endstops are not necessary.

bjela, 2018-05-15:
Just installed DRV4429 instead of LV8729 which I have been happily using quite some time.
I have some DRV8825, do you think they may help?
Funny enough DM452T's on NEMA23's are powered, but they are getting all the wires from RAMPS X driver slots.

What kind of line code do you suggest, I'll google it from there

Thanks

bjela, 2018-05-15:
Doing such a connection on ramps is kinda difficult. 
Isn't there other solution like in some 3d printer firmwares ?

bjela, 2018-05-15:
may it be related to the fact that there is no /joint_steps_feedback topic created
joint_steps and gripper angle are there on the other hand

bjela, 2018-05-15:
I thought that it was the drivers, so I just put 4 brand new drv4288. Didn't help.

Tomorrow I'll try new 8825s

I do control the base joint nema 23s and the servo trough the rviz, rosserial lounched folowed by convert and published gripper angle, all seems well except there is no feedback topic. 

Than you a lot

nez, 2018-10-05:
@ghost for the feedback topic, it looks like you have to uncomment the following line in the arduino code https://github.com/jesseweisberg/moveo_ros/blob/master/moveo_moveit/moveo_moveit_arduino/moveo_moveit_arduino.ino#L76

Did you manage to make the stepper motors move? and in that case, what drivers did you use? I have not been able to, using A4988 pololu

jesseweisberg, 2019-05-05:
Closing due to inactivity.

closed 2019-05-05 by jesseweisberg