jesseweisberg/moveo_ros #16: rviz crashing almost immediatley

MrDadaGuy, 2019-02-12:
Ubuntu 18.04, ROS Melodic...  First, thanks for putting this project out there, it's really fantastic!

When I start it up, it usually loads into rviz fine, but when I start moving the model (setting a Goal State), rviz crashes almost immediately.  Sometimes just when I click, sometimes when i'm moving the ball.  A couple of times it's been stable enough for me to create a goal pose and Plan & Execute a path.  But usually rviz crashes very quickly.

When I run a different model such as UR5, rviz is very stable and I've not experienced a crash despite a lot of use.  Any ideas, or suggestion on what I might try?  

Thanks,

Jeroenh89, 2019-02-20:
Hi,

You may want to try ROS Kinetic on Ubuntu 16.04 :)

MrDadaGuy, 2019-03-13:
Closing because I've rebooted since then and things seem ok now :)

closed 2019-03-13 by MrDadaGuy