jesseweisberg/moveo_ros #21: Can't get working in Gazebo

MrDadaGuy, 2019-03-14:
I wasn't able to get a gazebo render of the moveo, so I used moveit setup assistant to create a new moveit config directory.  This did create a demo_gazebo.launch file and give me an updated URDF containing gazebo meta data.
When I bring up rviz it looks good, but in gazebo all of the pieces are jumbled in a pile, not linked properly.  Any thoughts?  Thanks!

jesseweisberg, 2019-05-05:
This may have to do with the fact that the moments of inertia in the URDF's are not accurate and throwing off Gazebo.  The physics really have to be accurate for it to work in Gazebo, since it's a physics emulator, as opposed to just a visualization tool like RVIZ.

One solution would be to open the moveo model in a CAD program like SolidWorks that provides accurate moments for each link, and put those correct values into the URDF.

kgjeep, 2021-08-13:
this urdf works for me.
I have changed the inertias. But the control is not working. I ama working on it.

<?xml version='1.0'?>
<robot
  name="moveo_robot" xmlns:xacro="http://www.ros.org/wiki/xacro">	
  
<xacro:include filename="$(find moveo_urdf)/urdf/materials.xacro" />
<xacro:include filename="$(find moveo_urdf)/urdf/moveo.gazebo" />
     
     
  <link
    name="base_link">
    <visual>
      <origin
        xyz="0 0 .13"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://moveo_urdf/meshes/base_link.STL" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.75294 0.75294 0.75294 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 .13"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://moveo_urdf/meshes/base_link_col.STL" />
      </geometry>
    </collision>
    <inertial>
      <origin
        xyz="0.034155 -0.20591 -0.049641"
        rpy="0 0 0" />
      <mass
        value="8.0643" />
      
        <inertia
        ixx="0.446"
        ixy="0.0"
        ixz="0.0"
        iyy="0.446"
        iyz="0.0"
        izz="0.853" />
    </inertial>
   </link>

  <link
    name="odom">
</link>

  <joint
    name="odom_joint"
    type="fixed">
    <origin
      xyz="0 0 0"
      rpy="0 0 0" />
    <parent
      link="odom" />
    <child
      link="base_link" />
  </joint>

  <link
[... 695 lines cut ...]
      
      <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
    </joint>
    <actuator name="motor_third_forearm">
      
      <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

  <transmission name="forth_forearm_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="Joint_5">
      
      <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
    </joint>
    <actuator name="motor_forth_forearm">
      
      <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

  
</robot>