jesseweisberg/moveo_ros #24: Need advice to have ROS working.

dsolheid, 2019-06-10:
Hi, 

Since I didnt find the MATE distribution 14.0.4, I downloaded a complete distribution ROS + Ubuntu (for RPI 3+).

When I launch the demo you provided, I receive errors, and I think a made a mistake somewhere ..

Here the logs : 

`NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    move_group (moveit_ros_move_group/move_group)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz_ubiquityrobot_10451_2050894201 (rviz/rviz)
    virtual_joint_broadcaster_0 (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [10462]
ROS_MASTER_URI=http://192.168.0.27:11311

setting /run_id to e5cb4810-8bcf-11e9-a653-cf8a371fc380
process[rosout-1]: started with pid [10475]
started core service [/rosout]
process[virtual_joint_broadcaster_0-2]: started with pid [10492]
process[joint_state_publisher-3]: started with pid [10493]
process[robot_state_publisher-4]: started with pid [10500]
process[move_group-5]: started with pid [10504]
process[rviz_ubiquityrobot_10451_2050894201-6]: started with pid [10511]
[ INFO] [1560206069.607096443]: Loading robot model 'moveo_urdf'...
libEGL warning: DRI2: failed to authenticate
[ INFO] [1560206070.046252574]: rviz version 1.12.16
[ INFO] [1560206070.046545385]: compiled against Qt version 5.5.1
[ INFO] [1560206070.046649290]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1560206070.731920329]: Stereo is NOT SUPPORTED
[ INFO] [1560206070.732490585]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1560206071.382477814]: Loading robot model 'moveo_urdf'...
[move_group-5] process has died [pid 10504, exit code -11, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/ubuntu/.ros/log/e5cb4810-8bcf-11e9-a653-cf8a371fc380/move_group-5.log].
log file: /home/ubuntu/.ros/log/e5cb4810-8bcf-11e9-a653-cf8a371fc380/move_group-5*.log
[ INFO] [1560206075.969301223]: Loading robot model 'moveo_urdf'...
[ INFO] [1560206077.585081808]: Loading robot model 'moveo_urdf'...
[rviz_ubiquityrobot_10451_2050894201-6] process has died [pid 10511, exit code -11, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/ubuntu/catkin_ws/src/moveo_ros/moveo_moveit_config/launch/moveit.rviz __name:=rviz_ubiquityrobot_10451_2050894201 __log:=/home/ubuntu/.ros/log/e5cb4810-8bcf-11e9-a653-cf8a371fc380/rviz_ubiquityrobot_10451_2050894201-6.log].
log file: /home/ubuntu/.ros/log/e5cb4810-8bcf-11e9-a653-cf8a371fc380/rviz_ubiquityrobot_10451_2050894201-6*.log`

Thanks in advance if you can help me ,

Didier

dsolheid, 2020-06-30:
Hi,

One year later, I still try .. and same error. Exactly the same errors.

I reinstalled everything from scratch. Ubuntu 14.04, and so..

dsolheid, 2020-06-30:
```
[ INFO] [1593528444.895556735]: Loading robot model 'moveo_urdf'...
[rviz_rasppi_12253_176089222-6] process has died [pid 12308, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/moveo/catkin_ws/src/moveo_ros/moveo_moveit_config/launch/default.rviz __name:=rviz_rasppi_12253_176089222 __log:=/home/moveo/.ros/log/998ffdba-bae0-11ea-ba43-b827eb0495b7/rviz_rasppi_12253_176089222-6.log].
log file: /home/moveo/.ros/log/998ffdba-bae0-11ea-ba43-b827eb0495b7/rviz_rasppi_12253_176089222-6*.log

[ INFO] [1593528446.415788724]: Loading robot model 'moveo_urdf'...
[move_group-5] process has died [pid 12301, exit code -11, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/moveo/.ros/log/998ffdba-bae0-11ea-ba43-b827eb0495b7/move_group-5.log].
log file: /home/moveo/.ros/log/998ffdba-bae0-11ea-ba43-b827eb0495b7/move_group-5*.log
```

and the log doesnt exist ..

RVIZ open . .then die.

HUN-Adam8803, 2021-10-09:
The same problem here.
Did You find solution ?