jesseweisberg/moveo_ros #33: This is a problem "error: moveo_moveit/ArmJointState.h: No such file or directory" .

harunlakodla, 2021-02-24:
Hi ,

I did send a from e-mail this issues. I did make all command but it didn't run. continuous it give same bug. How can I do ? please...

I find devel/include/moveo_moveit/ArmStateJoint.h but it is not running. please....

harunlakodla, 2021-02-24:
I can do it...

SLINIS, 2021-08-10:
### I resolved the issue, take a look at: https://github.com/jesseweisberg/moveo_ros/issues/37#issuecomment-896127857

Tenveis, 2023-12-30:
You need to source the **moveo_ros** workspace before building library for arduino. 
Here is the link of  updated readme: [Tenveis:arduino_compile](https://github.com/Tenveis/moveo_ros/tree/arduino_compile)
You can find the instructions in **Troubleshoot** section.
Hope this helps.