jesseweisberg/moveo_ros #35: moveo_moveit/ArmJointState.h: No such file or directory. Tried everything in my current power. No luck

Bikerpilot1967, 2021-03-25:
Hi all,

I am trying to get moveo to work on Ubuntu 20.04 with ROS Noetic and am getting the "moveo_moveit/ArmJointState.h: No such file or directory" error. I tried all suggestions that I could find, but no luck. Manual copy of this file results in a cstring error, which, when resolved, causes a bunch of other issues, possibly related to the Eigen library. 

I even tried to revert back to Ubuntu 16 with ROS Kinetic. Same issue. Deleting ros_lib does not solve the issue. 

When I load the rosserial library (I tried 0.9.1 which causes an issue with cstring, I tried 0.7.8 and 0.7.9 which cause the ArmJointState error), I noticed that it actually installs a Rosserial_Arduino_Library instead of ros_lib. 

Is there anyone who has gotten this to work on Ubuntu 20.04 with ROS Noetic? I am at a loss and have been working on this on/off for days. I am sure I can figure it out after spending another few days, but any help would be appreciated. 

I have the simulation working, its going wrong at ros serial. For now my beautifully built robot arm sits aimlessly pointing at the ceiling....

Bikerpilot1967, 2021-03-25:
What the heck! Working on this for days, and I just decided to write the above issue, downloaded Rosserial 0.9.1 one more time and now it compiles!!! No idea why it works now.

warshoot3r, 2021-05-16:
heya buddy , how did you manage to fix it? im struggling with this at the moment.

SLINIS, 2021-08-10:
I resolve the issue: https://github.com/jesseweisberg/moveo_ros/issues/37#issuecomment-896127857

Tenveis, 2023-12-30:
You need to source the **moveo_ros** workspace before building library for arduino. 
Here is the link of  updated readme: [Tenveis:arduino_compile](https://github.com/Tenveis/moveo_ros/tree/arduino_compile)
You can find the instructions in **Troubleshoot** section.
Hope this helps.