jesseweisberg/moveo_ros #37: moveo_moveit/ArmJointState.h: No such file or directory

Relux-dev, 2021-05-26:
Basically for last month or so i try to work out this problem, I've tried everything that was written about this problem. I tried reinstalling the library many times. I even reinstalled my virtual machine few times. I tried everything that was written on the internet and nothing worked. I have the exact same problem that other folks do but i can't repair it, we are trying for weeks with my friend and it seems like it just doesn't work. Pls give us some advice, we are very eager to complete this robot and we just don't have a clue what to do anymore.

We use Ubuntu 16.04.07 and ros kinetic. We have everything working but rosserial library. It's the one thing that doesn't work.

Thank you in advice.

Kuber20, 2021-05-27:
I have the same problem and I don't know how to fix it?

kgjeep, 2021-08-04:
Try to do what it says in the Pull Request https://github.com/jesseweisberg/moveo_ros/pull/26 and after run CMake.
If the compile is ok then do what it says for the library.

SLINIS, 2021-08-10:
For me following approach resolved the issue (using ROS kinetic and Ubuntu 20.04.2) but it should  also work with ROS kinetic: 
**1. In the package.xml of moveo_moveit you need to add:** 

**1.1** `<build_depend>message_generation</build_depend>`
**1.2.** `<exec_depend>message_runtime</exec_depend> `
**1.3.**change `<run_depend>` to `<exec_depend>`  for all tags (don't forget to change   `</run_depend>` to ` </exec_depend>` as well ;) )
**1.4.**change ` <package> `to `<package format="2">`

**In the CMakeList.txt of moveo_moveit you need to change:** 
**2.1. add message_generation to the find_package function():**

```
find_package(catkin REQUIRED
  COMPONENTS
    roscpp
    rospy
    std_msgs
    moveit_core
    moveit_ros_planning
    moveit_ros_planning_interface
    pluginlib
    geometric_shapes
    moveit_visual_tools
    message_generation
)
```

**2.2 add message_runtime to the catkin_package() function:**
```
catkin_package(
  CATKIN_DEPENDS
    moveit_core
    moveit_ros_planning_interface
    interactive_markers
    message_runtime
  DEPENDS
    EIGEN3
)
```
Following tutorial helped me to resolve the issue: http://wiki.ros.org/ROS/Tutorials/CreatingMsgAndSrv#Creating_a_msg

_Unfortunately at the moment i have issues to attache the files (package.xml and CMakeList.txt). I will try to add them later._