jesseweisberg/moveo_ros #39: Electrical or Software problem with motors

AnOrdinaryUsser, 2022-01-29:
We are a group of students who are trying to replicate the construction of this robotic arm and we have encountered a problem that we are not able to solve.
We are using the code provided in this github repository, however we can't get the two motors at the bottom to work correctly with the code we have and we don't know very well how we have to configure everything to work perfectly and not sound a strange noise.
We have tried using the MultiStepperTest.ino code to test with these motors and have them spin correctly without skipping steps and without making strange noise but we have not been able to come up with a solution.

We are currently using these motors:
https://www.amazon.es/Edred%C3%B3n-pieza-23-Motor-alambre-drehbank/dp/B077XCHLN2/
And we have these two types of controllers:
https://www.amazon.es/DollaTek-bif%C3%A1sico-m%C3%A1x-Controlador-controlador-impresora/dp/B08HQCZ32N
And the TB6560

We have switched from using the TB6560 to the others because these first ones gave less amps than the motor indicated.
We also have the difficulty that inside the controllers there are some switches that we do not know exactly how to configure them, because we do not know what revolutions to use with these motors.

All the other motors work perfectly and these are the only ones that do not give us problems.

We need a little help with this issue, as we are not experts in this electrical section and it is almost our first contact with such large engines. If anyone can get in touch with me or be able to make a call, we would appreciate it.

Translated with www.DeepL.com/Translator (free version)

DominicM1213, 2022-02-12:
Ah yes. I am building my own at the moment and have already gone through your problem. This is my first for many things and working with steppers was another new thing I am trying. 

This is the website of the official Stepper Motor manufacturers: https://www.omc-stepperonline.com/nema-23-stepper-motor/e-series-nema-23-bipolar-1-8deg-3-0-nm-425oz-in-4-2a-57x57x113mm-4-wires.html. On it, you will get proper motor specifications and a helpful data sheet. Steps per revolution is calculated like this. 

For one full revolution, the standard motor has 200 steps. Divide that by 360 degrees, and you get 1.80 degrees per step. The motor requires 4.20 A to run properly. The TB6560 will not work as it is not rated for anything about 4 Amps. (I learned that the hard way) The controller you mentioned, along with a TB6600 would work because they can supply that amount of current. 

On the side there are switches called Dip Switches. On the top of your controller, it explains what each mode does. For the controller you sent, it looks like there should be 6 switches, 3 for microstepping and 3 for current. The 3 for current should be turned on or off to match the current of your motors. The other 3 microstep the motor. Microstepping basically increases the number of steps it takes to complete a revolution. Instead of 200 steps, maybe you want 400 steps, or even 6400 steps. The more steps, the smoother and quieter the motors, but the slower they move. I have mine at 8 microsteps for every one normal step for this project. I think Jessie mixed between 8 or 16. 

An issue I heard about was the potential to double the current. It is a practice when microstepping as the makeup of a stepper motor is made to be half on and half off, but when microstepping there is the chance when the motor is fully on, doubling the current.  I am doing this whole project with a 15 amp power supply, and I am doing just fine, but it doesn't hurt. I notice my robot moves differently when all motors run in unison.

xXhoobaXx, 2024-03-29:
i am facing the same issue 
i have my new motor and new driver, already configured the driver switched to 25 microsteps 
even tried different configurations and changed the conver file 

the reall issue i am having is that i am not getting the correct angle using any configuration 
i'd appreciate the help
also can anyone confirm that the small ramps drivers work or do i need to opt in for more big drivers?