jesseweisberg/moveo_ros #40: moveo_moveit/ArmJointState.h: No such file or directory

WilliamH07, 2022-02-22:
Hello,
I want to upload the arduino code with the arduino IDE but when I compile I have a error message : `moveo_moveit/ArmJointState.h: No such file or directory`.
 I try to find the file but it wasn't here, all I have is : ArmJointState.msg in the msg folder. I tried everything that I can find : I deleted the ros_lib, reinstall it,... I don't know at all why I have this but without the code I can't control the robot
Thanks 
WilliamH07

WilliamH07, 2022-03-06:
I try to reinstall Ubuntu but it didn't make any different, at this point I don't know what to do

TAZ010, 2022-07-08:
same issue here

usama0300854, 2023-11-05:
same error..any body knows how to fix this error?