jesseweisberg/moveo_ros #42: [ERROR] [1679484221.952115]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

anand-010, 2023-03-22:
Running on ubuntu 18.04 ros melodic.
```
rosrun rosserial_python serial_node.py /dev/ttyACM0
[INFO] [1679484144.807291]: ROS Serial Python Node
[INFO] [1679484144.814112]: Connecting to /dev/ttyACM0 at 57600 baud
[INFO] [1679484146.922112]: Requesting topics...
[ERROR] [1679484161.925963]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
```
this message repetes again and again.

anand-010, 2023-03-22:
any one what is the right version of ros_serial works with this repo?.

CKocher, 2024-03-05:
Hi, I know that it's been almost a year but did you find a solution to this problem? Any help would be highly appreciated! 

Best regards,

CK

closed 2024-03-06 by anand-010

CKocher, 2024-03-06:
So all you did was changing the ros_serial versions?

CKocher, 2024-03-06:
For anyone running into the same problem in the future: